fenlogic/vga666 #3: No output

Matty11611, 2015-02-06:
Hi, i like the idea of having two outputs from the Pi B+ however have bought the kit and tested all resistors are in the correct location followed the instructions several times from the manual and still no output, bought a new add on board and still nothing even after a fresh install and doing every step again. currently bought 2 add on boards with no success, please help, its part of my final year group project and i cant afford the time learning coding to work out where im going wrong. can i also add i found something about blacklisting which i have also done still with no joy, any other suggestions would be much appreciated, i have tried on 2 seprate vga monitors with different resolutions to no avail :(

gertvanloo, 2015-02-06:
Matty, I would love to help you but that is impossible. There are just too many ways in which it does not work. The first step towards a solution is to take an oscilloscope and check if you have signals on the pins, especially the hsync and vsync. It is also better to post your problem on the VGA section of Raspberry-Pi forum as a lot more people then just I alone can offer advise.

Matty11611, 2015-02-12:
Hi, been onto the rpi fourm and it seems to be an issue with the latest version of raspbian is there a fix coming soon for this to support the latest raspian release, thanks

Matty11611, 2015-02-14:
Gert, thanks for directing me over to RPI fourm, i have finally got it working, theres something wrong with the compatibilty of the VGA adapter with the last 2 versions of the raspbian, not sure if theres an extra step to be added or the dt-blob.bin file to be edited, also on your video it shows compatibilty for 2 screens, hdmi and VGA how can this be enabled, thanks

closed 2015-02-26 by gertvanloo