fenlogic/vga666 #4: Dual Screen

Matty11611, 2015-02-24:
Gert, just wondering if there is going to be any software available soon to enable both HDMI and VGA simultaneously, im currently trying to run impress with notes on the VGA screen with the os booting on vga and then get the ppt to run on the hdmi output, but dont really need it to be seen until the ppt is started, any ideas?

gertvanloo, 2015-02-26:
Matty, software to do that has been available from the beginning.
However as with all SW for the VGA adapter you have to ask the Raspberry-Pi Foundation if you are having any issues.

closed 2015-02-26 by gertvanloo

Matty11611, 2015-02-27:
I cant see s single place on there forum where you can download the software to make a global second monitor all I have found which forces dual screen using this is omxplayer, even using impress and a selection of display numbers still nothing even though the comand is of the same format as omxplayer

alkemir, 2016-09-30:
Dear Matty,
this might be a bit late, but check out https://github.com/alkemir/gertCloner
I don't know if we re-wrote the wheel here, but as you, I could not find a way to use
the VGA as a dual display. This is a step in that direction, not the solution you are asking
for, but maybe somebody will take it from there.

gertvanloo, 2017-02-12:
The latest answer I have from the foundation is here: https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=94424#p1104476

alkemir, 2017-02-12:
Hi Gert! Nice to hear from you. I will continue the discussion there,
hopefully Dom can assist us, I feel we are close to achieving extended
dualscreen for the RPi.