fenlogic/vga666 #5: setup/dt-blob-dpi.dts does not have references to the Pi2

robertely, 2015-03-19:
dt-blob-dpi.dts does not have a stanza for the Raspberry Pi 2. Preventing this from working. 

I've been trying to find a device tree with a video core config for the Pi 2 but I can't find enough information.

gertvanloo, 2015-03-19:
I do not have any control over the software. I know the VGA adapter was tested with the Pi-2 as I have seen it working with one of the production prototypes. As far as I know the drivers are part of the standard image. For all other matters post at "http://www.raspberrypi.org/forums/viewtopic.php?f=100&t=86658"

6by9, 2015-06-02:
Pi2 has used the bplus pins config until the firmware commit https://github.com/Hexxeh/rpi-firmware/commit/a1621d67b20f04acdac602d0b24923ec3cdbfb66 on 30th May.
Things have just changed - Phil is on it. (I've hit the issue when linking to a DPI screen via an Adafruit Kippah)

pelwell, 2015-06-03:
The new dt-blob has more configurations, each of which has a few more defines:
* rev1
* rev2
* bplus1 // Pi 1 Model B+ rev 1.1
* bplus2 // Pi 1 Model B+ rev 1.2
* aplus
* 2b1 // Pi 2 Model B rev 1.0
* 2b2 // Pi 2 Model B rev 1.1
* cm

The reason for the change has been to document the variations in hardware and to replace some of the board-specific code with additional usage of the pin configuration data.

It's a shame that the external dt-blob is a complete replacement for the internal one rather than a patch to it, but changing that isn't something to do in a hurry. Instead, we've uploaded the new structure [here](https://github.com/raspberrypi/documentation/blob/master/configuration/images/dt-blob.dts), and  you just need to overwrite pins 2-21 for the 5 (sorry) configurations for B+-compatible boards.