fenlogic/vga666 #7: Pi zero seems unsupported

fasmide, 2016-01-19:
Hi

Ive tried downloading your dt-blob.bin, putting it in /boot and added 

```
enable_dpi_lcd=1
display_default_lcd=1
dpi_group=2
dpi_mode=4
```
to `/boot/config.txt` - However im unable to find any V of H sync signals (with an oscilloscope) where they should be according to your diagram (Or at any other pin for that matter :)

Im thinking it could be related to the overlay not getting applied on the zero (I dont know how to debug this) but looking at the device tree source you provided - my guess is that it needs a "pin_zero" section, but im not sure what the pi zero is called though, looking in`/sys/firmware/devicetree/base`  does not get me anywhere:

```
pi@raspberrypi:/sys/firmware/devicetree/base $ cat name
pi@raspberrypi:/sys/firmware/devicetree/base $ cat model 
Raspberry Pi ? Rev 1.2
```

AndrewFromMelbourne, 2016-01-20:
Have a look at the (instructions on the Raspberry Pi forum)[https://www.raspberrypi.org/forums/viewtopic.php?p=657497#p657497], you don't need the dt-blob.bin any longer.

fasmide, 2016-01-20:
@AndrewFromMelbourne Thanks for pointing me in the right direction - ive managed to get it working :)

closed 2017-02-12 by gertvanloo