fenlogic/vga666 #8: Gerbers

qwertymodo, 2016-09-15:
I can't seem to load the gerber files in any of the gerber viewers I use, and OSHPark won't accept them either.  For one thing, most fab houses want their drill layer in Excellon format, and also, OSHPark requires a board outline on a separate layer.  I'm not sure what the issues with the other layers are, but I'm still looking into it, and I'll report back if I can figure it out.

qwertymodo, 2016-09-15:
Also, can you document what application should be used to open the .pcb file?  That file extension is used by a dozen different CAD applications, and I can't find any particularly useful information inside the file itself to tell me which one I should be using.

gertvanloo, 2017-05-13:
Pentalogix have a free gerber viewer (viewmate) which can handle these files as I use it myself.

closed 2017-05-13 by gertvanloo