fenlogic/vga666 #9: Tweak output of the VGA to other formats

vonj, 2016-10-03:
So, people, this got me thinking. Sorry if this is entirely the wrong place for this post. If so, any pointers welcome.

Anyhow, since it apparently is so easy (hardware wise, anyway) to make a VGA output from the Raspberry, can this approach be tweaked to do:

1)

15 kHz RGB output?  This way, Raspberry Pi could drive arcade and TV monitors for faithful emulation of old video games but with greater color separation than you get from the built in composite output.

2)
SVIDEO output. https://en.wikipedia.org/wiki/S-Video

3)
Composite video. (I know, maybe pointless considering the raspi already has one built in.)
https://en.wikipedia.org/wiki/Composite_video

4)
Drive the deflectors on an old black and white monitor/TV (X & Y) and video intensity directly?

I don't expect anyone to do these things for me, but would be very happy for any insights into how  this VGA thing is implemented on the Raspberry.

Where is all the magic happening? In the "bin" file? Where can I start tweaking settings and experiment with the output I get?

(For all of these output formats I am more interested in black/white output rather than color, if that helps.)

Antiriad, 2016-11-05:
Hi @vonj,
what you are asking in number 1 is yet doable with VGA 666:
[https://www.youtube.com/watch?v=gjUBQKyeWlQ](https://www.youtube.com/watch?v=gjUBQKyeWlQ)

And of course you can get a true RGB output on TV with a correct VGA - SCART cable. ;)

However, there seems to be an issue / limitation with the VGA 666 and pixel clock:
[https://www.raspberrypi.org/forums/viewtopic.php?p=1029349#p1029349](https://www.raspberrypi.org/forums/viewtopic.php?p=1029349#p1029349)

I really hope @gertvanloo  take a look at this post ...
Thanks! :)

vonj, 2016-11-11:
Thank you so much @Antiriad.

cedricp, 2018-08-29:
Hi,
Congratulation for this work. I'm interested in this project to drive an old CGA LCD monitor, but I would like to keep SPI pins free. I don't care losing color resolution, but I didn't find any documentation to setup the dts file.
Thank you.

gertvanloo, 2018-08-30:
No, the SPI interface uses GPIO 8-11 and those are the MS bits of the BLUE channel in all DPI modes. 
You can drop LS bits but the MS bit(s) are required. The minimum is 8 colours using only the MS bit of each channel.  
If you are desperate you can implement PIS using bit-banging.

closed 2018-08-30 by gertvanloo

cedricp, 2018-08-30:
Too bad ! I'll use a HW converter instead, thank you.