fenlogic/vga666 #13: Flickering with VGA666 on Raspbian Buster

alessioscand, 2019-10-07:
Hello,
the picture of VGA666 on Raspbian Buster has a flickering like this:
https://youtu.be/5f5hpGrPpZk

I've tested it on my RPi4 and I changed settings (resolution, refresh, video driver etc.) and even hardware (different GPIO cable, different VGA666) but the problem persist.
Thank you very much.

gertvanloo, 2019-10-07:
This is a known issue. It is mentioned in the assembly manual as "bit crawl". The cause is the cross-over talk between the various analogue signals. It gets worse with longer cables and higher frequencies. Some monitors are more susceptible then others.

closed 2019-10-07 by gertvanloo

alessioscand, 2019-10-07:
Thank you for your feedback. If this is the problem ("bitcrawl") why does it happen only on Raspbian Buster and/or RPi 4, and not on Raspbian Stretch / RPi 3?
I'm using same monitor and same parts.

linuxenko, 2020-03-07:
Hi @alessioscand. Did you make it work properly on RPI 4 ?