fenlogic/vga666 #14: VGA888 Information

TechieAndroid, 2020-10-17:
I believe there currently is a lack of organized information regarding how one would implement the additional resistors required for 8bits per channel. The 888 would require 6 extra resistors in total, 2 per channel.

If the current method is expanded upon I guess they would be 32K and 64K respectively?

Also since they're in parallel I'm guessing that they would connect to the same R/G/B channels respectively as the regular 666 resistors do. Then on the GPIO R would be pins 26&27 G pins 18&19 and B pins 6&7.

Am I right?

Also would the pi recognize it as a 888 vga hat or would one need to alter some part of the boot config as well? 

I just kinda pieced this information together from random parts of the vga manual so if anyone can clear this up I'd really appreciate it.

gertvanloo, 2020-10-17:
It is not just  a matter of adding more resistors. The 6 bit was chosen because of the tolerances of resistors. With 6 bits the tolerance has to be better then 1/64 (1.56% accuracy) which is do-able with 1% resistors. If not you get "colour banding". For 8 bit you would need 1/256 or better then 0.39%. Those are very difficult to obtain. Note that the accuracy is most critical in the larger resistor values.

TechieAndroid, 2020-10-17:
Thanks, that clears things up for me.

closed 2020-10-17 by TechieAndroid